ansys/pymotorcad #137: add a method to get the list of available template

gmalinve, 2023-06-21:
If it's possible it would be nice to have a method to get the list of all available templates and all their details.
Thanks!

jgsdavies, 2023-06-27:
The way Motor-CAD finds the template info is by searching the .mtt files in the templates folder. The files are plain text so just need to loop through and pull out:
'main_title'
Lots of extra info (type/drive type etc) in 'DataFile_Notes'

I think it would make sense to return this as a list of dictionaries e.g.
`[{'template': 'e9', 'winding type': 'stranded'}, {'template': 'i1', 'winding type': 'distributed'}]`

ravi-ansys, 2023-07-24:
@jgsdavies I may need a suggestion.   I have now the templates in dictionary format. i added under geometry .py file.  Check branch add_template_list. 
However i doubt it is not attached to Motor-CAD object. May you suggest where to add this list of template in code.

jgsdavies, 2023-07-25:
Hi Ravi, not sure this is the best way to do this since it won't update if the Motor-CAD templates change.

My suggestion is:

- Get file path of \templates folder. This will be under \Motor-CAD Data\templates. You can get the directory of the current Motor-CAD with: [image]
- Loop through all files in this folder with .mtt extension and retrieve the appropriate info directly from the .mot file e.g. 
[image]

ravi-ansys, 2023-07-25:
@jgsdavies  thanks for the tip. I wasn't aware about the command to get the installation director. 
Should i put the template list method (reading the .mtt files with Data_File_notes) under rpc_methos_geometry.py

jgsdavies, 2023-07-25:
@ravi-ansys Try writing this as a script first and then we can think about where it can live. The issue with rpc_methods_geometry.py (or any of the methods files) is that we don't have access to the other functions, e.g. get_variable