ansys/pymotorcad #139: Re-add error handling to _get_graph

jgsdavies, 2023-06-27:
### 📝 Description of the feature

[image]

ravi-ansys, 2023-07-21:
Hi Jack, 
              I have tried to use get_magnetic_graph which will use _get_graph() function. 
I modified it as below 

        except MotorCADError as e:
            # break
            # Can't be used currently: Bug in Motor-CAD RPC server
            if "Point requested is greater than number of points available" in str(e):
              
                break
            else:
                raise

however i get below popup and only get the plot when i press okay. Should this happen ? i think the if else statement here should work and popup should not appear on my screen 

[image]

jgsdavies, 2023-07-24:
I think this is due to Motor-CAD messagedisplaystate. Usually this is set to not show these messages when scripting.

Might be worth setting this to 2 while we do _get_graph and then resetting to previous setting.

ravi-ansys, 2023-07-27:
@jgsdavies 
 Hi, 
While setting the messagedisplaystate i am hitting the issue you mentioned in some other thread ("access to different method (set_variable" ) under rpc methods 
is there a way 
I tried this way which didn't work 

[image]

jgsdavies, 2023-08-15:
@ravi-ansys Sorry, didn't spot this before.

If you make```_get_graph``` a member of ```class _RpcMethodsGraphs``` then you can use ```self.set_variable("MessageDisplayState", 2)```

This works because ```_RpcMethodsGraphs``` is inherited by the main class.

If we are going to set MessageDisplayState then we need to make sure we revert back to the users original setting no matter what happens in our code. We can use a Try/Finally block to do this:
[image]

ravi-ansys, 2023-08-17:
@jgsdavies  I now placed the _get_graph under the _RPCMethodGraphs:
However i guess the following error saying we can't pass function as input arguments to _get_graph? 
[image]

closed 2024-01-09 by JackB-Ansys