ansys/pymotorcad #140: PyMotorCAD unable to launch new Motor-CAD instance

jgsdavies, 2023-07-11:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

When "Hide command Window" selected, PyMotorCAD can't launch new Motor-CAD.
[image]

The string in C:\Ansys_Motor-CAD\Shared Files\activex.bat changes format so `_find_motor_cad_exe()` in rpc_client_core.py fails to find the Motor-CAD exe. We need to handle this case.

### 📝 Steps to reproduce

- Select "Hide command Window" 
- Ensure registered version is current version
- Try launch Motor-CAD with `pymotorcad.MotorCAD()`

### 💻 Which operating system are you using?

Windows

### 🐍 Which Python version are you using?

3.11

### 📦 Installed packages

```shell
N/A
```

vpasagada, 2023-12-11:
Facing the same issue. Has this been resolved?

jgsdavies, 2023-12-14:
> Facing the same issue. Has this been resolved?

This has not been resolved yet but you can disable this option in Motor-CAD as it has no impact on behaviour when using PyMotorCAD