ansys/pymotorcad #170: Motor-CAD JSON-RPC server sending invalid JSON (e.g. `INF`)

vrzh, 2023-08-29:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

It seems reasonable that some Motor-CAD variables might take the value ∞. In which Motor-CAD seems to send the `INF` constant through the JSON-RPC interface. However, this is not valid JSON and a decode error is thrown on the Python side.

### 📝 Steps to reproduce

I believe I got into this situation when I stopped a solve early and then requested the value of the variable.

```
mcad.get_variable("IMSINGLELOADPOINT_POWERFACTOR")
```

Full stack trace:

```
Traceback (most recent call last):
  File "/lib/python3.10/site-packages/requests/models.py", line 971, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/lib/python3/dist-packages/simplejson/__init__.py", line 525, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 370, in decode
    obj, end = self.raw_decode(s)
  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 400, in raw_decode
    return self.scan_once(s, idx=_w(s, idx).end())
simplejson.errors.JSONDecodeError: Expecting value: line 1 column 58 (char 57)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/lib/python3.10/site-packages/ansys/motorcad/core/rpc_client_core.py", line 452, in send_and_receive
    response = requests.post(self._get_url(), json=payload).json()
  File "/lib/python3.10/site-packages/requests/models.py", line 975, in json
    raise RequestsJSONDecodeError(e.msg, e.doc, e.pos)
requests.exceptions.JSONDecodeError: Expecting value: line 1 column 58 (char 57)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "my-test.py", line 80, in <module>
    val = mcad.get_variable(v)
  File "/lib/python3.10/site-packages/ansys/motorcad/core/methods/rpc_methods_variables.py", line 67, in get_variable
    return self.connection.send_and_receive(method, params)
  File "/lib/python3.10/site-packages/ansys/motorcad/core/rpc_client_core.py", line 457, in send_and_receive
    self._raise_if_allowed("RPC Communication failed: " + str(e))
  File "/lib/python3.10/site-packages/ansys/motorcad/core/rpc_client_core.py", line 336, in _raise_if_allowed
    raise MotorCADError(error_message)
ansys.motorcad.core.rpc_client_core.MotorCADError: RPC Communication failed: Expecting value: line 1 column 58 (char 57)
```

Looking at the underlying JSON which is received:

```json
{"jsonrpc":"2.0","id":-1,"result":{"success":0,"output":[INF],"errorMessage":"pymotorcad: GetVariable IMSINGLELOADPOINT_POWERFACTOR = INF"}}
```

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

Ansys Motor-CAD v2023.2.1

### 🐍 Which Python version are you using?

3.10

### 📦 Installed packages

```shell
ansys-motorcad-core==0.2.0
```

vrzh, 2023-08-29:
Here is what it looks like in the Motor-CAD UI:

[image]

Not sure what the best solution is, a change in MotorCAD to not send the `INF` constant, or somehow relax the JSON parsing on the Python side. For example, the [`simplejson`](https://simplejson.readthedocs.io/en/latest/#simplejson.loads) has `allow_nan` and `parse_constant` which could be useful.

jgsdavies, 2023-09-13:
Thanks for spotting this @vrzh. We will look into fixing this for a future release. Also feel free to submit a PR if you have a good solution already.

vrzh, 2023-09-14:
Not sure what is the best solution, as it seems there is no JSON standard to represent these types of values.

One option is to change Motor-CAD to output `Infinity` instead of `INF` as this seems to be what Python JSON decoders typically accept:

```py
# python3 -m pip install 'simplejson==3.19.1'

import simplejson
simplejson.loads('Infinity', allow_nan=True)

# python3 -m pip install 'simplejson<3.19.1'

import simplejson
simplejson.loads('Infinity')

# using built-in json package
import json
json.loads('Infinity')
```

It looks like `pymotorcad` uses the `requests` library. But, the problem is that `requests` library only uses `simplejson` if it is installed, otherwise it falls back to `json`. 

Therefore `pymotorcad` would also have to check if `simplejson` is installed and what version it is (adding `allow_nan=True` for older versions of `simplejson` will raise an exception). Or, add `simplejson` as a dependency of `pymotorcad` with a fixed version (either the old or new one).

jgsdavies, 2023-11-28:
Should be fixed for Motor-CAD 2024R2. Returns "None" instead of throwing exception

closed 2024-09-17 by jgsdavies