ansys/pymotorcad #221: Use myst-nb for example documentation.

dcrawforAtAnsys, 2023-11-08:
### 📝 Description of the feature

Could we please use [myst-nb](https://myst-nb.readthedocs.io/en/latest/) for the notebook examples?  This will greatly simplify creation and debugging of the notebooks by enabling real-time feedback while editing.

### 💡 Steps for implementing the feature

See https://myst-nb.readthedocs.io/en/latest/

jorgepiloto, 2023-11-10:
Thanks for suggesting this, @dcrawforAtAnsys. I think it is a great idea and a nice opportunity. More than happy to implement this.

MaxJPRey, 2023-11-16:
Adding @SMoraisAnsys .

dcrawforAtAnsys, 2023-11-22:
I believe the best solution would be to use sphinx format with *.py files with the following header:
```
# ---
# jupyter:
#   jupytext:
#     text_representation:
#       extension: .py
#       format_name: sphinx
#       format_version: '1.1'
#       jupytext_version: 1.14.5
#   kernelspec:
#     display_name: Python 3 (ipykernel)
#     language: python
#     name: python3
# ---
```
This makes it possible to view/edit notebooks in Jupyter using the jupytext plugin with minimal change to the current CI/CD.

dcrawforAtAnsys, 2023-12-04:
It's still not clear that myst-nb is the best path for the creation of examples. We will continue to investigate.

MaxJPRey, 2023-12-08:
@dcrawforAtAnsys , I think it could be really useful and adapted in your case.