ansys/pymotorcad #248: Add corner rounding functions to Region() class

jgsdavies, 2024-01-09:
### 📝 Description of the feature

Create Region class methods to apply corner rounding.

[image]

```python
region = mc.get_region("stator")

# Apply corner rounding to a specific point
# Raise exception if not possible
region.round_corner(int(region_point), float(radius))

# Apply corner rounding to whole region
# Raise exception if not possible
region.round_region(float(radius))

# Function to get max radius of a point
region.get_max_corner_radius(int(region_point))
```

### 💡 Steps for implementing the feature

Probably best to create the actual functionality in the ```EntityList``` class and then link to the new ```EntityList``` methods in ```Region```

Start with 2 lines for now. Is it possible/sensible to do line-arc or arc-arc corners?

jgsdavies, 2024-01-09:
@james-packer Can you add any use cases that you would have for this.

jmaravin, 2024-09-06:
I will be quite interested if this function gets implemented. 
thanks for your effort.

closed 2024-11-27 by jgsdavies