ansys/pymotorcad #253: Bug located in Region.material - cannot set material

JackB-Ansys, 2024-01-23:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

The script does not set Rotor material to "M470-50A" as it should. Material remains set to original material "M250-35A".

Script set in Geometry -> Editor -> Adaptive Templates tab in the attached Motor-CAD file (in .zip).
[e9_eMobility_IPM_User.zip](https://github.com/ansys/pymotorcad/files/14023614/e9_eMobility_IPM_User.zip)

### 📝 Steps to reproduce

Open attached Motor-CAD file (in .zip) in Motor-CAD v2024R1 
Go to Geometry -> Editor -> Adaptive Templates
Click 'Run'
Observe Material of Rotor is still set to original value.

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

Motor-CAD 2024R1

### 🐍 Which Python version are you using?

3.10

### 📦 Installed packages

```shell
-e git+https://github.com/pyansys/pymotorcad/@47b79e89706f6152d60c9116e329de5923252ff5#egg=ansys_motorcad_core
attrs==22.2.0
cachetools==5.2.1
certifi==2022.12.7
cfgv==3.3.1
chardet==5.1.0
charset-normalizer==3.0.1
click==8.1.3
colorama==0.4.6
contourpy==1.0.7
cycler==0.11.0
distlib==0.3.6
exceptiongroup==1.1.0
filelock==3.9.0
Flask==2.2.2
fonttools==4.39.0
identify==2.5.13
idna==3.4
importlib-metadata==6.0.0
iniconfig==2.0.0
itsdangerous==2.1.2
Jinja2==3.1.2
jsonrpcserver==5.0.9
jsonschema==4.17.3
kiwisolver==1.4.4
MarkupSafe==2.1.2
matplotlib==3.7.1
nodeenv==1.7.0
numpy==1.24.2
OSlash==0.6.3
packaging==23.0
Pillow==9.4.0
platformdirs==2.6.2
pluggy==1.0.0
pre-commit==2.21.0
psutil==5.9.4
pyparsing==3.0.9
pyproject_api==1.5.0
pyrsistent==0.19.3
pytest==7.2.1
python-dateutil==2.8.2
PyYAML==6.0
requests==2.28.2
six==1.16.0
tomli==2.0.1
tox==4.3.5
typing_extensions==4.5.0
urllib3==1.26.14
virtualenv==20.17.1
Werkzeug==2.2.2
zipp==3.11.0
```

yyang262, 2024-08-29:
will this bug be fixed? i have similar issue when assigning a magnet material (Vinyl Ferrite) to a region. looks like it's still behaving like air.

JackB-Ansys, 2025-09-09:
@pyansys-automation never close issue

This issue is still not fixed in v2026R1 test release