ansys/pymotorcad #387: Add method to check whether region crosses into airgap

JackB-Ansys, 2024-09-25:
### 📝 Description of the feature

Add functionality to PyMotorCAD to be able to detect the case where a region crosses the rotor/stator boundary into airgap

jgsdavies, 2024-10-17:
@matthewAnsys is this fixed with the latest Motor-CAD release geom checks?

EmilCallerfjord, 2025-01-21:
Hello @jgsdavies , @matthewAnsys. 

Any updates regarding this feature?

How would I go on and implement this geometry airgap check when using OptiSLang? Currently I have created a script where I iterate the coordinates in my entities and if they cross the rotor diameter in polar coordinates an error gets raised. But it is still possible to run simulations. 

I was thinking of somehow writing "success = 0" in the optiSlang script when my geometry enters the airgap but I do not know how to use the adaptive template python script inside MotorCAD together with the Optimization Script created for optiSLang. 

Best regards,
Emil

JackB-Ansys, 2025-09-09:
Hi Emil, 

If you have built this check into your Adaptive Templates script, I recommend that you raise an exception when the your check is failed.

Then, if you add a Geometry Check into your optimisation script, Motor-CAD will run the geometry check and this will fail due to the Exception raised by the Adaptive Templates script.

I realise this reply is late but I hope it is helpful in case you did not solve the issue in the meantime.

JackB-Ansys, 2025-09-09:
This may be possible with the **fix_duct_geometry** method introduced with #545 

Needs further investigation