ansys/pymotorcad #450: Runtime Error : maximum handles for the current process is met

abdsharaf, 2025-01-06:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

Hi,

I have started to run an FEA simulation with MotorCAD using Python. The maximum number of simulations that are expected to be run is 40000 simulations. for each iteration. a new motor geometry is generated which is used for getting a new e-magnetic calculation. However, a few errors occur during the simulations, which unfortunately cannot be continued to the end.

The errors are related to the maximum allowed handles for the actual process which is is met. 

from the python console it seems that the it could only 467 have been successfully excuted and the results are merged to a CSV file, and around 9980 simulations have been initialized.

Error during simulation8312: process no longer exists (pid=3972)Error during simulation8308: process no longer exists (pid=11488)

Error during simulation8311: process no longer exists (pid=23160)

Starting initialization for model 8320.
Starting initialization for model 8318.Starting initialization for model 8316.Starting initialization for model 8317.

### 📝 Steps to reproduce

    with concurrent.futures.ThreadPoolExecutor(max_workers=10) as executor:
        future_to_row = {}
        for _, row in df.iterrows():
            # Submit the task to the executor
            future = executor.submit(process_row, row)
            future_to_row[future] = row

            # Introduce a 1-second delay before submitting the next task
            time.sleep(2)
        for future in concurrent.futures.as_completed(future_to_row):
            row = future_to_row[future]
            print('future_to_row')
            try:
                inputs, result = future.result()
                if result:
                    completed_count += 1
                    print(f"Simulation {completed_count}/{total_rows} completed")

                    if completed_count % merge_interval == 0:
                        batch_num += 1
                        merge_csv_files(batch_num, merge_interval, 'TrainDataBase.csv')
            except Exception as exc:
                print(f"Simulation failed for inputs: {inputs}. Error: {exc}")

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

MotorCAD 2024_2_2.2

### 🐍 Which Python version are you using?

3.11

### 📦 Installed packages

```shell
ansys-motorcad-core
```

jgsdavies, 2025-01-08:
Hi @abdsharaf, are you opening a new Motor-CAD for each calculation?

abdsharaf, 2025-01-08:
@jgsdavies , for each working task i open a Motor-CAD for one design geomaery and after the calculation is finished, it will be quit and a new GUI will be opened, so that the max_workers of tasks is 10 ( the parallel number of MotorCAD GUIs running)

jgsdavies, 2025-06-30:
Hi @abdsharaf, You are probably running into the limit of parallel instances of Motor-CAD that can be open at a given time. We are improving this with every Motor-CAD release.

closed 2025-06-30 by jgsdavies