ansys/pymotorcad #455: Issue with Exporting Lab Model elecdata as MATLAB-Compatible .mat File in Motor-CAD

Ahmedabdelrahman88, 2025-01-07:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

Hi there, 

I am trying to export the Lab model elecdata results from Motor-CAD using the Python API, specifically to generate a MATLAB-compatible .mat file similar to the MotorLAB_elecdata.mat file provided in the Motor-CAD GUI exports.

Here’s what I have tried:

Script Execution:

I configured the Lab settings (e.g., SpeedMax, SpeedInc, SpeedMin) and successfully ran the calculate_magnetic_lab() method.
I used export_results("Lab", <output_path>) to export the results. However, the generated .mat file is not MATLAB-compatible and does not contain the structured elecdata results.
I attempted to parse the text data manually and recreate the .mat structure, but it does not replicate the expected elecdata structure from the GUI-exported MotorLAB_elecdata.mat.
Export Attempt Using export_lab_model:

I tried using the export_lab_model(<output_path>) method, but it either produces errors (e.g., method not found) or does not output the expected data.

Issue:

I need to extract and export the Lab model elecdata results into a .mat file that is identical in structure and content to the MotorLAB_elecdata.mat file generated via the Motor-CAD GUI.

Could you confirm the correct method to export the Lab model elecdata results using the Python API?

Regards, 
Ahmed[Simulation_script_Modified.zip](https://github.com/user-attachments/files/18335553/Simulation_script_Modified.zip)

### 📝 Steps to reproduce

Load the Motor-CAD Project:

Open Motor-CAD or use the Python API to load the .mot file:
javascript
Copy code
C:/------------------/PMMotor5V8_090C.mot
Configure Lab Settings:

Set the following Lab configuration parameters via the Python API:
python
Copy code
mcApp.set_variable('EmagneticCalcType_Lab', 0)  # Magnetic calculation type
mcApp.set_variable('SpeedMax_MotorLAB', 15000)  # Maximum speed
mcApp.set_variable('Speedinc_MotorLAB', 250)    # Speed increment
mcApp.set_variable('SpeedMin_MotorLAB', 250)    # Minimum speed
Run the Magnetic Lab Calculation:

Use the Python API to calculate the magnetic Lab results:
python
Copy code
mcApp.calculate_magnetic_lab()
Attempt to Export Lab Model Results:

Export the Lab model results using the following methods:
Method 1:
python
mcApp.export_results("Lab", "C:/path_to_results/Lab_Results.mat")

The exported file is generated but is not a valid MATLAB .mat file and does not contain the required elecdata structure.

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

2024.2.3.1

### 🐍 Which Python version are you using?

3.13

### 📦 Installed packages

```shell
ansys-motorcad-core==0.6.4
anyio==4.6.2.post1
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
arrow==1.3.0
asttokens==2.4.1
async-lru==2.0.4
attrs==24.2.0
babel==2.16.0
beautifulsoup4==4.12.3
bleach==6.2.0
certifi==2024.8.30
cffi==1.17.1
charset-normalizer==3.4.0
colorama==0.4.6
comm==0.2.2
contourpy==1.3.1
cycler==0.12.1
debugpy==1.8.8
decorator==5.1.1
defusedxml==0.7.1
executing==2.1.0
fastjsonschema==2.20.0
fonttools==4.54.1
fqdn==1.5.1
h11==0.14.0
httpcore==1.0.6
httpx==0.27.2
idna==3.10
importlib_metadata==8.5.0
ipykernel==6.29.5
ipython==8.29.0
ipywidgets==8.1.5
isoduration==20.11.0
jedi==0.19.2
Jinja2==3.1.4
json5==0.9.28
jsonpointer==3.0.0
jsonschema==4.23.0
jsonschema-specifications==2024.10.1
jupyter==1.1.1
jupyter-console==6.6.3
jupyter-events==0.10.0
jupyter-lsp==2.2.5
jupyter_client==8.6.3
jupyter_core==5.7.2
jupyter_server==2.14.2
jupyter_server_terminals==0.5.3
jupyterlab==4.2.5
jupyterlab_pygments==0.3.0
jupyterlab_server==2.27.3
jupyterlab_widgets==3.0.13
kiwisolver==1.4.7
MarkupSafe==3.0.2
matplotlib==3.9.2
matplotlib-inline==0.1.7
mistune==3.0.2
nbclient==0.10.0
nbconvert==7.16.4
nbformat==5.10.4
nest-asyncio==1.6.0
networkx==3.4.2
notebook==7.2.2
notebook_shim==0.2.4
numpy==2.1.3
overrides==7.7.0
packaging==24.2
pandas==2.2.3
pandocfilters==1.5.1
parso==0.8.4
pillow==11.0.0
platformdirs==4.3.6
prometheus_client==0.21.0
prompt_toolkit==3.0.48
psutil==6.1.0
pure_eval==0.2.3
pycparser==2.22
Pygments==2.18.0
pyparsing==3.2.0
python-dateutil==2.9.0.post0
python-json-logger==2.0.7
pytz==2024.2
pywin32==308
pywinpty==2.0.14
PyYAML==6.0.2
pyzmq==26.2.0
referencing==0.35.1
requests==2.32.3
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rpds-py==0.21.0
scipy==1.14.1
Send2Trash==1.8.3
setuptools==75.4.0
six==1.16.0
sniffio==1.3.1
soupsieve==2.6
stack-data==0.6.3
terminado==0.18.1
tinycss2==1.4.0
tornado==6.4.1
traitlets==5.14.3
types-python-dateutil==2.9.0.20241003
tzdata==2024.2
uri-template==1.3.0
urllib3==2.2.3
wcwidth==0.2.13
webcolors==24.11.1
webencodings==0.5.1
websocket-client==1.8.0
widgetsnbextension==4.0.13
zipp==3.21.0
```

james-packer, 2025-01-08:
Hello, the MotorLAB_elecdata.mat file is automatically generated when you run the lab calculation, so it should already exist (so you do not need to call any methods to export it explicitly).

To find where it is located, you can use something like the following

```
mcApp = pymotorcad.MotorCAD()
mcApp.set_variable('EmagneticCalcType_Lab', 0) # Magnetic calculation type
mcApp.set_variable('SpeedMax_MotorLAB', 15000) # Maximum speed
mcApp.set_variable('Speedinc_MotorLAB', 250) # Speed increment
mcApp.set_variable('SpeedMin_MotorLAB', 250) # Minimum speed
mcApp.calculate_magnetic_lab()
results_path = mcApp.get_variable('ResultsPath_MotorLab') + 'MotorLAB_elecdata.mat'
# Do something with the .mat file
```

Please check if this does what is needed.

Ahmedabdelrahman88, 2025-01-09:
Thank you @james-packer. I figured it out.

closed 2025-01-28 by james-packer