ansys/pymotorcad #456: Setting up a winding scheme via Python in MotorCAD

vsa-add, 2025-01-09:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

Following is python script for creating winding schemes in MotorCAD:

**Input:**
A reference winding scheme is defined in seperate py script "define_winding_scheme.py" in form of dictionary. Functions defined in following below script are used to create multiple winding schemes and save them into seperate ."mot" files.

**Issue**
However, the problem is that the script is not able to set the winding scheme nor opening the MotorCAD file. 

**Python script:** 

import ansys.motorcad.core as mcd
import define_winding_scheme

mcdApp = mcd.MotorCAD()

# Function to open the .mot file
def open_file(file_path):
    try:
        mcdApp.load_from_file(file_path)
        print(f"Successfully opened {file_path}")
    except Exception as e:
        print(f"Error opening {file_path}: {e}")

# Function to shift the winding of a given phase by a certain number of slots
def shift_phase_winding(phase_winding, shift_by, max_slots):
    shifted_winding = {}

    # Iterate over both paths (Path 1 and Path 2)
    for path, windings in phase_winding.items():
        shifted_winding[path] = []

        # Iterate over each winding tuple
        for winding in windings:
            start_slot, start_layer, end_slot, end_layer = winding

            # Shift the start and end slots by the given shift value
            shifted_start_slot = (start_slot + shift_by - 1) % max_slots + 1  # Ensure 1-based index
            shifted_end_slot = (end_slot + shift_by - 1) % max_slots + 1  # Ensure 1-based index

            # Append the shifted winding tuple
            shifted_winding[path].append((shifted_start_slot, start_layer, shifted_end_slot, end_layer))

    return shifted_winding

# Create a function to set the winding scheme using MotorCAD API
def define_coil_path(phase, path, coil, go_slot, go_position, return_slot, return_position, turns):
    # Call MotorCAD API to set the winding coil
    mcdApp.set_winding_coil(
        phase,                # Phase number
        path,                 # Path number
        coil,                 # Coil number
        go_slot,              # Go slot (first element in the dataset)
        go_position,          # Go position (second element in the dataset)
        return_slot,          # Return slot (third element in the dataset)
        return_position,      # Return position (fourth element in the dataset)
        turns                 # Number of turns, assumed to be 1
    )

# Function to iterate through the winding data and set the winding scheme
def configure_winding(winding_data):
    for phase, paths in winding_data.items():  # Iterate through each phase
        for path, coils in paths.items():  # Iterate through each path in a phase

            for coil_idx, coil_data in enumerate(coils, start=1):  # Iterate through each coil
                go_slot, go_position, return_slot, return_position = coil_data  # Extract coil data
                # Call the MotorCAD function with phase, path, coil number, and coil data
                define_coil_path(
                    phase=phase,
                    path=path,
                    coil=coil_idx,
                    go_slot=go_slot,
                    go_position=go_position,
                    return_slot=return_slot,
                    return_position=return_position,
                    turns=1  # Assuming turns=1 as per your initial request
                )

# Function to shift any phase in the winding data
def shift_phase_in_winding(winding_data, phase_to_shift, shift_by, max_slots):
    # Check if the specified phase exists in the winding data
    if phase_to_shift in winding_data:
        # Shift the specified phase
        shifted_phase_winding = shift_phase_winding(winding_data[phase_to_shift], shift_by, max_slots)

        # Update the winding data with the shifted phase
        winding_data[phase_to_shift] = shifted_phase_winding
    else:
        print(f"Phase {phase_to_shift} does not exist in the winding data.")

# Main script execution

if __name__ == "__main__":

    mcd_file_path = r'C:\JDA_0\generate_WindingScheme.mot'

    winding_scheme = define_winding_scheme.data_set

    shift_phase_in_winding(winding_scheme, phase_to_shift=1, shift_by=0, max_slots=72)

    configure_winding(winding_scheme)

    updated_mcd_file_path = r'C:\JDA_0\generate_WindingScheme.mot'

    mcdApp.save_to_file(updated_mcd_file_path)

    print(f"Updated MotorCAD file saved as {updated_mcd_file_path}")`

### 📝 Steps to reproduce

However the code shows following error!

### Error: 

Traceback (most recent call last):
  File "C:\Users\VedantSane\PycharmProjects\MotorCAD\shifting_method.py", line 90, in <module>
    configure_winding(winding_scheme)
    ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
  File "C:\Users\VedantSane\PycharmProjects\MotorCAD\shifting_method.py", line 57, in configure_winding
    define_coil_path(
    ~~~~~~~~~~~~~~~~^
        phase=phase,
        ^^^^^^^^^^^^
    ...<6 lines>...
        turns=1  # Assuming turns=1 as per your initial request
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "C:\Users\VedantSane\PycharmProjects\MotorCAD\shifting_method.py", line 38, in define_coil_path
    mcdApp.set_winding_coil(
    ~~~~~~~~~~~~~~~~~~~~~~~^
        phase,                # Phase number
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<6 lines>...
        turns                 # Number of turns, assumed to be 1
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "C:\Users\VedantSane\PycharmProjects\MotorCAD\.venv\Lib\site-packages\ansys\motorcad\core\methods\rpc_methods_geometry.py", line 65, in set_winding_coil
    return self.connection.send_and_receive(method, params)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
  File "C:\Users\VedantSane\PycharmProjects\MotorCAD\.venv\Lib\site-packages\ansys\motorcad\core\rpc_client_core.py", line 565, in send_and_receive
    self._raise_if_allowed(error_message)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "C:\Users\VedantSane\PycharmProjects\MotorCAD\.venv\Lib\site-packages\ansys\motorcad\core\rpc_client_core.py", line 387, in _raise_if_allowed
    raise MotorCADError(error_message)
ansys.motorcad.core.rpc_client_core.MotorCADError: pymotorcad: SetWindingCoil
pymotorcad: Coil index too high. Check number of coils per path.
pymotorcad: SetWindingCoil failed

Process finished with exit code 1

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

Anysys MotorCAD 2024.1.1

### 🐍 Which Python version are you using?

3.9

### 📦 Installed packages

```shell
ansys-motorcad-core==0.6.1
certifi==2024.8.30
charset-normalizer==3.4.0
idna==3.10
importlib_metadata==8.5.0
packaging==24.1
psutil==6.0.0
requests==2.32.3
urllib3==2.2.3
zipp==3.20.2
```

vsa-add, 2025-01-09:
**Input winding scheme defined in define_winding_scheme.py**

data_set = {
    1: {  # Phase 1
        1: [  # Path 1
            (3, "h", 9, "g"),
            (15, "h", 21, "g"),
            (27, "h", 33, "g"),
            (39, "h", 45, "g"),
            (51, "h", 57, "g"),
            (63, "h", 69, "g"),
            (3, "f", 9, "e"),
            (15, "f", 21, "e"),
            (27, "f", 33, "e"),
            (39, "f", 45, "e"),
            (51, "f", 57, "e"),
            (63, "f", 69, "e"),
            (3, "d", 9, "c"),
            (15, "d", 21, "c"),
            (27, "d", 33, "c"),
            (39, "d", 45, "c"),
            (51, "d", 57, "c"),
            (63, "d", 69, "c"),
            (3, "b", 9, "a"),
            (15, "b", 21, "a"),
            (27, "b", 33, "a"),
            (39, "b", 45, "a"),
            (51, "b", 57, "a"),
            (63, "b", 69, "a"),
            (63, "a", 57, "b"),
            (51, "a", 45, "b"),
            (39, "a", 33, "b"),
            (27, "a", 21, "b"),
            (15, "a", 9, "b"),
            (3, "a", 69, "b"),
            (63, "c", 57, "d"),
            (51, "c", 45, "d"),
            (39, "c", 33, "d"),
            (27, "c", 21, "d"),
            (15, "c", 9, "d"),
            (3, "c", 69, "d"),
            (63, "e", 57, "f"),
            (51, "e", 45, "f"),
            (39, "e", 33, "f"),
            (27, "e", 21, "f"),
            (15, "e", 9, "f"),
            (3, "e", 69, "f"),
            (63, "g", 57, "h"),
            (51, "g", 45, "h"),
            (39, "g", 33, "h"),
            (27, "g", 21, "h"),
            (15, "g", 9, "h"),
            (3, "g", 69, "h")
        ],
        2: [  # Path 2
            (4, "h", 10, "g"),
            (16, "h", 22, "g"),
            (28, "h", 34, "g"),
            (40, "h", 46, "g"),
            (52, "h", 58, "g"),
            (64, "h", 70, "g"),
            (4, "f", 10, "e"),
            (16, "f", 22, "e"),
            (28, "f", 34, "e"),
            (40, "f", 46, "e"),
            (52, "f", 58, "e"),
            (64, "f", 70, "e"),
            (4, "d", 10, "c"),
            (16, "d", 22, "c"),
            (28, "d", 34, "c"),
[... 217 lines cut ...]
            (48, "a", 42, "b"),
            (36, "a", 30, "b"),
            (24, "a", 18, "b"),
            (12, "c", 6, "d"),
            (72, "c", 66, "d"),
            (60, "c", 54, "d"),
            (48, "c", 42, "d"),
            (36, "c", 30, "d"),
            (24, "c", 18, "d"),
            (12, "c", 6, "d"),
            (72, "e", 66, "f"),
            (60, "e", 54, "f"),
            (48, "e", 42, "f"),
            (36, "e", 30, "f"),
            (24, "e", 18, "f"),
            (12, "e", 6, "f"),
            (72, "g", 66, "h"),
            (60, "g", 54, "h"),
            (48, "g", 42, "h"),
            (36, "g", 30, "h"),
            (24, "g", 18, "h"),
            (12, "g", 6, "h"),
        ]
    }
}

RobPasMue, 2025-01-27:
@jgsdavies @JackB-Ansys @james-packer @matthewAnsys - could you please help out this user on his request?

james-packer, 2025-01-28:
@vsa-add, can I check if you have selected a hairpin winding with the parallel paths to 2 and winding layers to 8 (which I think matches the data you have shown above)? You can either set this in the Motor-CAD GUI, or through PyMotorCAD with mcdApp.set_variable('ParallelPaths_Hairpin', 2) and mcdApp.set_variable('WindingLayers', 8)

When I set up an example motor with 72 slots, hairpin winding, 8 layers, 2 parallel paths, your script seems to run as expected.

vsa-add, 2025-01-28:
@james-packer Thanks for your reply! I think 2 parallel paths and 8 winding layers are set already. May I ask if you ran the script outside MotorCAD environment? It works if I run it internally through MotorCAD scripting. However, I am using Pycharm to run it externally. Please let me know if we could set up a short meeting to review the code. Thanks!

james-packer, 2025-01-28:
@vsa-add, in this case, can I check if your script calls your 'open_file' function? I can see the definition in your code, but it doesn't seem to be called anywhere.

vsa-add, 2025-01-28:
@james-packer Ah yes! That's what I was missing. Now it works ;)

closed 2025-01-28 by vsa-add