ansys/pymotorcad #464: Add ...Winding - Definition - Redraw function

mhss22, 2025-01-29:
### 📝 Description of the feature

Can you add a winding-definition-recalculate function, please 
This should execute the code that is run by the Redraw button on the winding, definition tab

(underlying issue is that the NoMushConductorsDrawn variable is not updated when the fill factor "GrossSlotFillFactor" 
is changed in pymotorcad) 

### 💡 Steps for implementing the feature

add calculate-redraw function to calculations

james-packer, 2025-01-29:
Hi @mhss22. If you call the `do_weight_calculation()` function after changing the RequestedGrossSlotFillFactor variable, this will trigger the calculations required, and NoMushConductorsDrawn will be updated.

mhss22, 2025-02-01:
Thanks James - I can confirm this works 
I had tried check_if_geometry_is_valid(0) but had not thought to do the weight check 
Regards 
Malcolm

mhss22, 2025-03-05:
I think (am sure) there is a bug in this approach 

If Armature_Winding_Definition is set to 0 or 1 the ConductorsPerSlot and NoMushConductorsDrawn are not updated following a weight calculation - however, these variables are updated when Armature_Winding_Definition is set to 2 (but this calc uses the wrong wire diameter)

Put the following in any script 

    wire_diameter = mcad.get_variable("Wire_Diameter")
    mcad.set_variable("Armature_Winding_Definition", 0) #change the 0 to 1 and 2
    for i in range(1, 5):

        no_strands_in_hand = i
        mcad.set_variable("NumberStrandsHand", no_strands_in_hand)
        # weight calc updates the Mush
        mcad.do_weight_calculation()
        # check the drawing
        no_strands_in_hand = mcad.get_variable("NumberStrandsHand")
        no_conductors_per_slot = mcad.get_variable("ConductorsPerSlot")
        no_conductors_per_slot_drawn = mcad.get_variable("NoMushConductorsDrawn")
        wire_diameter = mcad.get_variable("Wire_Diameter")
        print(
            f"Strands {no_strands_in_hand} cond per slot {no_conductors_per_slot} drawn {no_conductors_per_slot_drawn }"
        )