ansys/pymotorcad #486: EStringListError when attempting to close MotorCad Instance on certain files.

mwilljohnson, 2025-03-07:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

On some MotorCad files when you attempt to exit via `MotorCAD.quit()` the following error box pop's up even if message boxes are suppressed. This will lock up python until the user clicks okay. 

[image]

Text of the error:

`Exception EStringListError in module rtl270.bpl at 0000.... List index out of bounds (0).`

Other operations can be performed on files which have this error before attempting to exit without issue. Only certain files exhibit this behavior. I am not sure what in the file would cause this to happen though.

I am using python 3.8.20 and MotorCad version 2024 1.3.1.

The error can be bypassed if you do not run in blackbox mode and call `MotorCad.set_visible(True)` before calling `MotorCAD.quit()`. This functionality does not work in blackbox mode.

### 📝 Steps to reproduce

The following code is enough to cause certain files to crash. I have attached two motorcad files `21s7p_frontConfig_finless_base.mot` will produce the error, `24s4p_CoDR_motor_1_base.mot` runs as expected.
```
import ansys.motorcad.core as pymotorcad

mcad = pymotorcad.MotorCAD()
mcad.set_variable('MessageDisplayState', 2)
basefile = 'C:/Users/martin.johnson/Documents/GitHub/Whisper_eMach/21s7p_test.mot'
mcad.load_from_file(basefile)
mcad.quit()

```

[BugReportFiles.zip](https://github.com/user-attachments/files/19132638/BugReportFiles.zip)

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

MotorCad version 2024 1.3.1.

### 🐍 Which Python version are you using?

3.9

### 📦 Installed packages

```shell
alabaster @ file:///home/conda/feedstock_root/build_artifacts/alabaster_1673645646525/work
ansys-motorcad-core==0.7.0
arrow @ file:///home/conda/feedstock_root/build_artifacts/arrow_1696128962909/work
astroid @ file:///D:/bld/astroid_1721516296397/work
asttokens @ file:///home/conda/feedstock_root/build_artifacts/asttokens_1698341106958/work
atomicwrites @ file:///home/conda/feedstock_root/build_artifacts/atomicwrites_1657325823582/work
attrs @ file:///home/conda/feedstock_root/build_artifacts/attrs_1722977137225/work
autopep8 @ file:///home/conda/feedstock_root/build_artifacts/autopep8_1693061251004/work
babel @ file:///home/conda/feedstock_root/build_artifacts/babel_1730878832677/work
backcall @ file:///home/conda/feedstock_root/build_artifacts/backcall_1592338393461/work
backports.tarfile @ file:///home/conda/feedstock_root/build_artifacts/backports.tarfile_1730878908799/work
bcrypt @ file:///D:/bld/bcrypt_1721690127380/work
beautifulsoup4 @ file:///home/conda/feedstock_root/build_artifacts/beautifulsoup4_1705564648255/work
binaryornot==0.4.4
black @ file:///D:/bld/black-recipe_1723489059187/work
bleach @ file:///home/conda/feedstock_root/build_artifacts/bleach_1696630167146/work
Brotli @ file:///D:/bld/brotli-split_1666788538925/work
certifi @ file:///home/conda/feedstock_root/build_artifacts/certifi_1725278078093/work/certifi
[... 138 lines cut ...]
stack-data @ file:///home/conda/feedstock_root/build_artifacts/stack_data_1669632077133/work
text-unidecode @ file:///home/conda/feedstock_root/build_artifacts/text-unidecode_1694707102786/work
textdistance @ file:///home/conda/feedstock_root/build_artifacts/textdistance_1721228410193/work
three-merge @ file:///home/conda/feedstock_root/build_artifacts/three-merge_1595515817927/work
tinycss2 @ file:///home/conda/feedstock_root/build_artifacts/tinycss2_1729802851396/work
toml @ file:///home/conda/feedstock_root/build_artifacts/toml_1604308577558/work
tomli @ file:///home/conda/feedstock_root/build_artifacts/tomli_1727974628237/work
tomlkit @ file:///home/conda/feedstock_root/build_artifacts/tomlkit_1723631480657/work
tornado @ file:///D:/bld/tornado_1717722870081/work
tqdm @ file:///home/conda/feedstock_root/build_artifacts/tqdm_1730926118411/work
traitlets @ file:///home/conda/feedstock_root/build_artifacts/traitlets_1713535121073/work
types-python-dateutil @ file:///home/conda/feedstock_root/build_artifacts/types-python-dateutil_1727940235703/work
typing_extensions @ file:///home/conda/feedstock_root/build_artifacts/typing_extensions_1717802530399/work
ujson @ file:///D:/bld/ujson_1715783259235/work
unicodedata2 @ file:///D:/bld/unicodedata2_1695848058846/work
urllib3 @ file:///home/conda/feedstock_root/build_artifacts/urllib3_1726496430923/work
watchdog @ file:///D:/bld/watchdog_1716561935652/work
wcwidth @ file:///home/conda/feedstock_root/build_artifacts/wcwidth_1704731205417/work
webencodings @ file:///home/conda/feedstock_root/build_artifacts/webencodings_1694681268211/work
whatthepatch @ file:///home/conda/feedstock_root/build_artifacts/whatthepatch_1720999218266/work
win_inet_pton @ file:///D:/bld/win_inet_pton_1727796272493/work
yapf @ file:///home/conda/feedstock_root/build_artifacts/yapf_1690387939953/work
zipp @ file:///home/conda/feedstock_root/build_artifacts/zipp_1726248574750/work
zstandard==0.23.0
```

mwilljohnson, 2025-03-07:
This somehow seems to be connected to if potting is used in the model. By turning that on/off I can change the behavior of a file (cause a good file to crash and a bad file to run normally).  This behavior seems to vary computer to computer, some co-workers were able to run file without issue, some were not.

JackB-Ansys, 2026-10-09:
This issue could not be reproduced on our end.

closed 2026-10-09 by JackB-Ansys