ansys/pymotorcad #494: RotorOuterDiameter variable of BPMOR rotor does not seem to work

jmaravin, 2025-04-01:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

I am trying to set the variable "RotorOuterDiameter" in BPMOR motors through the following command and does not seem to work. 

mcad.set_variable("RotorOuterDiameter", 160)  

I´ve realized on the GUI, this variable has a bracket specifying that is a dimension and it is the only variable which has this kind of further explanation. See picture below. I wonder if this variable is of a different type and its value is not set though the common .set_variable code. 

[image]

### 📝 Steps to reproduce

Open an existing BPMOR motor and run the following code in your virtual environment. 

    mcad = pymotorcad.MotorCAD()    
    mcad.show_magnetic_context()
    mcad.display_screen("Scripting")
    mcad.set_variable("GeometryParameterisation", 1)
    mcad.set_variable("RotorOuterDiameter", i_ROD)  

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

V2025R1

### 🐍 Which Python version are you using?

3.11

### 📦 Installed packages

```shell
annotated-types==0.7.0
ansys-motorcad-core==0.7.0
ansys-pythonnet==3.1.0rc4
attrs==23.2.0
branca==0.7.2
certifi==2024.2.2
cffi==1.16.0
charset-normalizer==3.3.2
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
contourpy==1.2.0
customtkinter==5.2.2
cycler==0.12.1
darkdetect==0.8.0
defusedxml==0.7.1
et-xmlfile==1.1.0
fiona==1.9.5
folium==0.16.0
fonttools==4.49.0
fpdf2==2.7.8
geopandas==0.14.3
idna==3.6
imageio==2.34.0
importlib-metadata==7.0.1
Jinja2==3.1.3
jsonschema==4.21.1
jsonschema-specifications==2023.12.1
kiwisolver==1.4.5
MarkupSafe==2.1.5
matplotlib==3.8.4
motulator==0.1.7
networkx==3.2.1
numpy==1.26.0
openpyxl==3.1.2
osmnx==1.9.1
packaging==23.2
pandas==2.1.1
pillow==10.2.0
platformdirs==4.2.0
plumbum==1.8.2
pooch==1.8.0
psutil==5.9.8
pycparser==2.21
pydantic==2.9.2
pydantic_core==2.23.4
pyedb==0.34.3
pyparsing==3.1.1
pyproj==3.6.1
python-dateutil==2.8.2
pytomlpp==1.0.13
pytz==2024.1
pyvista==0.42.2
pywin32==306
PyYAML==6.0.2
referencing==0.33.0
requests==2.31.0
rpds-py==0.18.0
rpyc==6.0.0
Rtree==1.3.0
schemdraw==0.18
scikit-rf==0.31.0
scipy==1.12.0
scooby==0.9.2
shapely==2.0.3
six==1.16.0
SRTM.py==0.3.7
toml==0.10.2
tomli_w==1.1.0
typing_extensions==4.12.2
tzdata==2024.1
urllib3==2.2.0
utm==0.7.0
vtk==9.2.6
xyzservices==2024.4.0
zipp==3.17.0
```

james-packer, 2025-04-22:
@jmaravin please could you show the error that you get? For me, the script you shared seems to run correctly, both when run inside Motor-CAD and running PyMotorCAD externally. In my test, I set i_ROD = 200:

[image]

jmaravin, 2025-05-27:
@james-packer thanks very much for your message.. I found out the error I was having and it was due to the fact I was trying to set the Rotor Outer Diameter and the Rotor back iron thickness at the same time. The former using mcad.set_variable("GeometryParameterisation", 1) and the latter with mcad.set_variable("GeometryParameterisation", 0). However, once I changed one, the other one was automatically changed as they are related. 

It is now sorted so I will proceed to close the issue.

closed 2025-05-27 by jmaravin