ansys/pymotorcad #557: (Again) keep_instance_open is not available

OkayasuShinsuke, 2025-07-30:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

[r2_User.zip](https://github.com/user-attachments/files/21504449/r2_User.zip)

I made a script which has open mot file, and set geometry dimension, do magnetic calculation.

I would like to check parameters on Motor-CAD gui after magnetic calculation.
but the motor-cad instance is closed automatically after execute the final line in the script.

Why is the keep instance option unavalable?
Is there some mistake in my script? (See attached file) 

### 📝 Steps to reproduce

・Set mot file and .py file in the same directory.
・change directory in the attached script following your environment.
・run the script
・Probably the motorcad instance will be closed after saving new mot file.

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

Motor-CAD 25R2

### 🐍 Which Python version are you using?

3.9

### 📦 Installed packages

```shell
about-time==4.2.1
aiohappyeyeballs==2.4.0
aiohttp==3.10.5
aiosignal==1.3.1
alive-progress==3.1.5
altair==5.5.0
annotated-types==0.7.0
ansys-acp-core==0.1b2
ansys-additive-core==0.18.1
ansys-additive-widgets==0.2.1
ansys-api-acp==0.1.0.dev9
ansys-api-additive==1.7.2
ansys-api-dbu==0.3.4
ansys-api-dyna==0.3.6
ansys-api-edb==1.0.6
ansys-api-fluent==0.3.27
ansys-api-geometry==0.4.6
ansys-api-mapdl==0.5.2
ansys-api-mechanical==0.1.2
ansys-api-meshing-prime==0.1.3
ansys-api-modelcenter==0.3.1
ansys-api-platform-instancemanagement==1.1.0
ansys-api-pyensight==0.4.1
ansys-api-sherlock==0.1.27
ansys-api-systemcoupling==0.1.0
ansys-api-tools-filetransfer==0.1.0
ansys-api-workbench==0.2.0
ansys-conceptev-core==0.1.4
ansys-dpf-composites==0.5.0
ansys-dpf-core==0.12.2
ansys-dpf-post==0.8.0
ansys-dyna-core==0.4.15
ansys-dynamicreporting-core==0.6.0
ansys-edb-core==0.1.6
ansys-engineeringworkflow-api==0.1.0
ansys-fluent-core==0.22.0
ansys-geometry-core==0.6.6
ansys-grantami-bomanalytics==2.1.1
ansys-grantami-bomanalytics-openapi==3.0.0
ansys-grantami-jobqueue==1.0.1
ansys-grantami-recordlists==1.2.1
ansys-grantami-serverapi-openapi==3.0.0
ansys-hps-client==0.8.0
ansys-mapdl-core==0.68.4
ansys-mapdl-reader==0.54.1
ansys-math-core==0.1.5
ansys-mechanical-core==0.11.5
ansys-mechanical-env==0.1.7
ansys-meshing-prime==0.6.2
ansys-modelcenter-workflow==0.1.1
ansys-motorcad-core==0.5.5
ansys-openapi-common==2.1.1
ansys-optislang-core==0.8.0
ansys-platform-instancemanagement==1.1.2
ansys-pyensight-core==0.8.8
ansys-pythonnet==3.1.0rc4
ansys-rocky-core==0.2.0
ansys-seascape==0.2.0
ansys-sherlock-core==0.6.0
ansys-simai-core==0.2.2
ansys-sound-core==0.1.3
ansys-systemcoupling-core==0.6
ansys-tools-filetransfer==0.1.0
ansys-tools-local-product-launcher==0.1.0
ansys-tools-path==0.6.0
ansys-tools-visualization-interface==0.4.3
ansys-turbogrid-api==0.4.3
ansys-turbogrid-core==0.4.1
ansys-units==0.3.2
ansys-workbench-core==0.6.0
anyio==4.4.0
appdirs==1.4.4
asgiref==3.8.1
asttokens==3.0.0
async-timeout==4.0.3
attrs==24.2.0
azure-core==1.32.0
azure-identity==1.19.0
backoff==2.2.1
backports.entry-points-selectable==1.3.0
backports.tarfile==1.2.0
bcrypt==4.2.0
beartype==0.18.5
beautifulsoup4==4.12.3
bleach==6.1.0
blinker==1.8.2
blosc2==2.7.1
bokeh==3.4.3
build==1.2.2
cachetools==5.5.0
certifi==2025.4.26
cffi==1.15.1
charset-normalizer==3.3.2
click==8.1.7
clr-loader==0.2.6
colorama==0.4.6
comm==0.2.2
contourpy==1.3.0
cryptography==43.0.1
cycler==0.11.0
dataclasses-json==0.6.7
debugpy==1.8.9
decorator==5.1.1
defusedxml==0.7.1
Deprecated==1.2.14
dill==0.3.8
dirtyjson==1.0.8
distro==1.9.0
Django==5.1.1
docker==7.1.0
docutils==0.21.2
EasyProcess==1.1
elementpath==4.4.0
entrypoint2==1.1
et-xmlfile==1.1.0
exceptiongroup==1.2.2
executing==2.1.0
fabric==3.2.2
fast_simplification==0.1.7
filelock==3.16.0
filetype==1.2.0
Flask==3.0.3
flexcache==0.3
flexparser==0.3.1
fonttools==4.42.1
fpdf2==2.7.9
frozenlist==1.4.1
fsspec==2024.10.0
geomdl==5.3.1
geopandas==1.0.1
gitdb==4.0.11
GitPython==3.1.43
google-api-core==2.19.2
google-api-python-client==2.144.0
google-auth==2.34.0
google-auth-httplib2==0.2.0
googleapis-common-protos==1.65.0
grapheme==0.6.0
greenlet==3.1.1
grpcio==1.66.1
grpcio-health-checking==1.48.2
grpcio-status==1.48.2
h11==0.14.0
httpcore==1.0.5
httplib2==0.22.0
httpx==0.26.0
idna==3.4
imageio==2.36.1
importlib-metadata==6.8.0
invoke==2.2.0
ipykernel==6.29.5
ipython==8.30.0
itsdangerous==2.2.0
jaraco.classes==3.4.0
jaraco.context==6.0.1
jaraco.functools==4.0.2
jedi==0.19.2
Jinja2==3.1.4
jiter==0.8.0
joblib==1.4.2
jsonpatch==1.33
jsonpointer==3.0.0
jsonschema==4.23.0
jsonschema-specifications==2023.12.1
jupyter_client==8.6.3
jupyter_core==5.7.2
keyring==25.3.0
kiwisolver==1.4.5
langchain==0.3.8
langchain-core==0.3.21
langchain-text-splitters==0.3.2
langsmith==0.1.146
linkify-it-py==2.0.3
llama-cloud==0.1.5
llama-index==0.12.2
llama-index-agent-openai==0.4.0
llama-index-cli==0.4.0
llama-index-core==0.12.2
llama-index-embeddings-azure-openai==0.3.0
llama-index-embeddings-openai==0.3.0
llama-index-indices-managed-llama-cloud==0.6.2
llama-index-legacy==0.9.48.post4
llama-index-llms-azure-openai==0.3.0
llama-index-llms-openai==0.3.2
llama-index-multi-modal-llms-openai==0.3.0
llama-index-program-openai==0.3.1
llama-index-question-gen-openai==0.3.0
llama-index-readers-file==0.4.0
llama-index-readers-llama-parse==0.4.0
llama-parse==0.5.15
lxml==5.3.0
Markdown==3.7
markdown-it-py==3.0.0
MarkupSafe==2.1.5
marshmallow==3.22.0
marshmallow-oneofschema==3.1.1
matplotlib==3.9.2
matplotlib-inline==0.1.7
mdit-py-plugins==0.4.1
mdurl==0.1.2
meshio==5.3.5
more-itertools==10.5.0
MouseInfo==0.1.3
msal==1.31.1
msal-extensions==1.2.0
msgpack==1.0.8
mss==9.0.1
multidict==6.0.5
mypy-extensions==1.0.0
narwhals==1.14.2
ndindex==1.9.2
nest-asyncio==1.6.0
networkx==3.3
nh3==0.2.18
nltk==3.9.1
numexpr==2.10.2
numpy==1.25.2
openai==1.57.0
openpyxl==3.1.2
orjson==3.10.12
osmnx==2.0.0
outcome==1.3.0.post0
overrides==7.7.0
packaging==24.2
pandas==2.1.0
panel==1.4.4
param==2.1.1
paramiko==3.4.1
parso==0.8.4
Pillow==10.0.0
Pint==0.24.3
pkginfo==1.10.0
platformdirs==4.2.2
plumbum==1.8.2
pooch==1.8.2
portalocker==2.10.1
prompt_toolkit==3.0.48
proto-plus==1.24.0
protobuf==3.20.3
psutil==5.9.5
pure_eval==0.2.3
py-cpuinfo==9.0.0
pyaedt==0.12.0
pyansys==2024.2.4
pyansys-tools-variableinterop==0.1.1
pyansys-tools-versioning==0.6.0
pyarrow==18.1.0
pyasn1==0.6.0
pyasn1_modules==0.4.0
PyAutoGUI==0.9.54
pycparser==2.21
pydantic==2.8.2
pydantic_core==2.20.1
pydeck==0.9.1
pyedb==0.34.3
PyGetWindow==0.0.9
Pygments==2.18.0
pygranta==2024.2.0
pyiges==0.3.1
PyJWT==2.9.0
PyMsgBox==1.0.9
PyNaCl==1.5.0
pyogrio==0.10.0
pyparsing==3.1.1
pypdf==5.1.0
pyperclip==1.8.2
pypiwin32==223
pypng==0.20220715.0
pyproj==3.7.0
pyproject_hooks==1.1.0
PyRect==0.2.0
Pyro5==5.15
pyscreenshot==3.1
PyScreeze==0.1.29
PySocks==1.7.1
pyspnego==0.11.1
python-dateutil==2.9.0.post0
python-dotenv==1.0.1
python-pptx==1.0.2
pythonnet==3.0.2
pytomlpp==1.0.13
pytweening==1.0.7
pytwin==0.7.0
pytz==2023.3.post1
pyvista==0.43.4
pyviz_comms==3.0.3
pywin32==306
pywin32-ctypes==0.2.3
PyYAML==6.0.2
pyzmq==26.2.0
rdp==0.8
readme_renderer==44.0
referencing==0.35.1
regex==2024.7.24
requests==2.32.3
requests-negotiate-sspi==0.5.2
requests-toolbelt==1.0.0
requests_ntlm==1.3.0
rfc3986==2.0.0
rich==13.8.0
rpds-py==0.20.0
rpyc==6.0.0
rsa==4.9
Rtree==1.3.0
schedule==1.2.2
scikit-learn==1.5.2
scikit-rf==1.3.0
scipy==1.11.3
scooby==0.10.0
scriptenv==0.3.0
selenium==4.33.0
semver==3.0.2
serpent==1.41
shapely==2.0.6
six==1.16.0
smmap==5.0.1
sniffio==1.3.1
sortedcontainers==2.4.0
soupsieve==2.5
SQLAlchemy==2.0.36
sqlparse==0.5.1
SRTM.py==0.3.7
sseclient-py==1.8.0
sspilib==0.1.0
stack-data==0.6.3
streamlit==1.40.2
striprtf==0.0.26
tables==3.10.1
tabulate==0.9.0
tenacity==8.5.0
threadpoolctl==3.5.0
tiktoken==0.8.0
toml==0.10.2
tomli==2.0.1
tomli_w==1.1.0
tornado==6.4.1
tqdm==4.66.5
traitlets==5.14.3
trame==3.6.3
trame-client==3.2.5
trame-server==3.1.2
trame-vtk==2.8.10
trame-vuetify==2.7.1
trio==0.30.0
trio-websocket==0.12.2
twine==5.1.1
typing-inspect==0.9.0
typing_extensions==4.13.2
tzdata==2023.3
uc-micro-py==1.0.3
uritemplate==4.1.1
urllib3==2.4.0
utm==0.7.0
vtk==9.3.1
wakepy==0.9.1
watchdog==6.0.0
wcwidth==0.2.13
webencodings==0.5.1
websocket-client==1.8.0
websockets==13.0.1
Werkzeug==3.0.4
WMI==1.5.1
wrapt==1.16.0
wslink==2.1.2
wsproto==1.2.0
XlsxWriter==3.2.0
xmlschema==2.5.1
xyzservices==2024.9.0
yarl==1.11.0
zipp==3.17.0
```

JackB-Ansys, 2025-08-05:
Hi,
From your list of installed packages, it seems you have `ansys-motorcad-core==0.5.5`. Please update to 0.7.4 and try again.

When I try running your script, Motor-CAD stays open as expected. I tried this in the current release 25R2, and in the 26R1 test release I have. I did not try 25R2SP1 which I do not have installed. Can you try in 25R2 and see whether it works? 
Thanks!

ShinsukeOkayasu, 2025-09-10:
Hi, @JackB-Ansys 

Sorry for my very late reply.
I checked it in 25R1 and 25R2SP1. (I have currently these two versions.)
version is pymotorcad 0.7.4, python 3.12.4
The automation is worked. but the keep open instance is not worked.
Should I download 26R1 (developed version)?

https://github.com/user-attachments/assets/d1f37280-6809-41a9-a9a8-5714c7614e99

[test_mot.zip](https://github.com/user-attachments/files/22245764/test_mot.zip)